3DChameleon/3DChameleonMk4 #1: Simplify electronics ordering

davidedelvento, 2024-12-03:
Hi,
Thanks for making this project public (rather than "kill it" when you felt it wasn't your priority anymore). I wish I found it earlier so I could have purchased from you instead, but I digress....

I see in the electronics folder that you used EasyEDA for the design. Rather than having us all to "reinvent the wheel" and make sure we correctly upload all files, BOMs etc, do you mind clicking on just one button to make that happen automatically? 

Just share your project from EasyEDA, as described at https://docs.easyeda.com/en/Introduction/Share-to-Public/ 

If you instead used PCBWay, the button to click to get the same result is described at https://www.pcbway.com/blog/help_center/How_to_Share_Projects_in_PCBWay_Community_d27959dc.html

Of course people can still grab the files from github and upload them to their favorite vendor if they feel so inclined...

Thanks again for opening the project and for considering this suggestion.

PS: if anybody is taking the baton and offering for sale a kit of parts for sale, please comment here! I'm interested especially the electronics, since most people can find the hardware in any hardware store and print the 3d-printed ones on their own...

wjsteele, 2024-12-03:
I'll argue the point that I felt it wasn't my priority anymore, I've never said that anywhere.  My priority is to get this into the hands of everybody that wants it and selling/manufacturing it was getting in my way of supporting/expanding it.  We have new projects that we also want to work on and I simply didn't have all the time I needed for all of them... this was clearly the best route to take to allow that.

As for the design documents, you can simply import the full project file (which is included in the files) into EasyEDA... right now, their system won't allow me to share it publicly for some reason, and despite your link to their instructions, they simply don't work, which is why I posted them here as a full project file.  It's not difficult and there is no hunting... you open the json file and you get everything linked.

Importing it brings in 100% of the entire project, including the BOM and links their part numbers.  It also includes the second Mini Mk4 board that allows one of the new methods of installation at an even lower cost.

Bill

closed 2024-12-03 by wjsteele

wjsteele, 2024-12-11:
Instead of using the stock electronics, you can also use an Arduino CNC Shield or the Arduino CNC Nano shields in exactly the same manner.  In our firmware, you can simply change the pin mappings to your required boards.  You'll get 100% the same functionality from them and they're very inexpensive.

wjsteele, 2024-12-11:
There are no current sensing capabilities in the existing designs, nor were there any planned.  They only need either the AS5600 motor position sensor, which is a $1 hall effect rotation sensor using the I2C bus, or simple filament position sensors, using generic "limit" switches to detect the position.  The CNC shield actually is a much cheaper solution and includes more functionality... as the original electronics needed the I/O expansion to support this, but these shields can do with without any expansions needed.

tyeth, 2024-12-12:
Thanks for the clarification, much appreciated!