3DChameleon/3DChameleonMk4 #5: Inclusion of .Step file for CAD

lordgitface, 2025-01-04:
HI  was wondering if it's possible to include .step file of the CAD as it would seem the 'send to fusion' option in tinkerCAD seems not to work .

Cheers.

wjsteele, 2025-01-04:
No, it is not possible... the files were created in Tinkercad.  There are no other source files.  You can export them from Tinkercad as OBJ or STL files and convert them.  Both Fusion 360 and FreeCad can create STEP files from the STL files.

closed 2025-01-04 by wjsteele

tyeth, 2025-01-17:
I found the tinkerCAD model to be too complex to export to fusion (another export option), it always failed to export to Fusion, but after removing two of the autoClippys and leaving the third + blade holder the export completed successfully. I can now use fusion to export into whatever I like (e.g. step).

[image]