3DChameleon/3DChameleonMk4 #7: Not an issue, just a big thanks for sharing your design

cotestatnt, 2025-01-31:
First of all, thank you for sharing your design!

I successfully built and installed a slightly modified version of the 3D Chameleon color changer on a Bambulab A1, and it works really well. 
I can complete a color change in less than 20 seconds because the filament path on the A1 is very short. There is still room to improve this further, for example, by increasing the stepper motor speed.

The only aspect that didn’t fully convince me is the small teeth system for securing the PTFE tubes. So, I adapted your design to accommodate these fittings that I had available, cut in half:

[image]

I believe there is also enough space to screw in a standard PC4-M6 fitting without any major issues.

For the electronics, I repurposed some components I had on hand, using a CNC shield for an Arduino Nano and a couple of Hall sensors instead of microswitches. I also added a sensor for homing the selector without bringing the motor to a mechanical stop, as I found it to be somewhat inaccurate—the motor tends to "bounce" when it starts skipping steps. While this isn’t a major issue, since I had the sensor available, I preferred this approach.
As for the firmware, since speed is more important to me than flexibility or the max number of possible colors, I simplified the color selection process, as I usually only need 2–3 colors for my designs.

primesun, 2025-03-12:
@cotestatnt Would you be willing to share a photo of the overall setup? I'm thinking of trying this out and it would be helpful.

cotestatnt, 2025-03-13:
@primesun Sure!
As soon as I can, I'll take some photos of my setup.
I can also share the modified parts and firmware sources if you're interested.

primesun, 2025-03-14:
@cotestatnt If you could, I would greatly appreciate it!

mcmatrix, 2025-08-04:
> Hi [@primesun](https://github.com/primesun) Here are the modified parts (Part B and Part C) as STEP files, along with my custom firmware. I've also included some pictures of my current setup, as well as a video demonstrating a manual selection. Apologies for the "floating" pushbutton; it was added after I had already printed the electronics box because it is more convenient than using the hall effect sensor when it's needed.
> 
> https://github.com/cotestatnt/3DChameleonMk4
> 
> Regarding the firmware, all the libraries used are included in the project as local files. I prefer this approach, as it ensures they won't interfere with any other libraries you might have installed on your system.

Thank you for sharing your solution.
I haven't found any detailed information about a solution made for either the A1 or A1 Mini.
I've been thinking for a long time about adding a 3DCHameleon to my Bambu A1 Mini, but I see you've made it for the A1. How stable does it work? Did you place the switch near the cutting mechanism with a Hall sensor and magnet? That seems like a clever location.
Could you post more pictures of your setup?