3DChameleon/3DChameleonMk4 #9: Electronics BOM has incorrect capacitor part

TotalGriffLock, 2025-03-31:
Hi,

In the process of making my 3D Chameleon, I have had a run of PCBs made and placed the BOM order with LCSC. The capacitors that are in the BOM for C5 and C6 are this part number:

"6"	"100uF"	"C5,C6"	"CAP-SMD_BD8.0-L8.3-W8.3-FD"	"2"	"CA45-B-6.3V-100uF-K"	"CEC"	"LCSC"	"C129289"	"0.107"

https://www.lcsc.com/product-detail/Tantalum-Capacitors_CEC-Shenzhen-Zhenhua-XinYun-Elec-CA45-B-6-3V-100uF-K_C129289.html

What turns up is your usual tiny square SMD type component, yet the PCB is clearly designed to take a radial can type capacitor.

wjsteele, 2025-03-31:
This is a known issue with LCSC.  The part linked, however, is actually a part that will work and fit.  It's not tiny... it's the same size as the one it replaces and fits the same pads.

Bill

TotalGriffLock, 2025-03-31:
That's cool, as long as it is a known issue 👍 was just a bit of a surprise!

closed 2025-03-31 by TotalGriffLock