3DChameleon/3DChameleonMk4 #10: Tikercad files linked from this Github are in poor condition

Maker-Paul, 2025-04-01:
My previous issue was closed without proper consideration.  The concept of 3Dchameleon is brilliant but badly let down by the state of the Tinkercad files linked from the front page of this Github.

I have never heard of "Tinkercad's iterative method" which was given as the reason for the 100's of embedded copies within the Tinkercad parts.

What has actually happened is multiple people have hit the Duplicate and repeat button and not separated the model for editing many many times. The result is files that are now a nightmare to edit and if used as they are the holes don't even line up. I know it is a remix and I know the Tinkercad files are not stored on this Github but you are providing a link to something which I'm sure you would not wish to represent what is otherwise a very good and exciting project. I have cleaned the files up and used the mirror function to properly produce the top and bottom halves of the parts with all hole lining up if they are of use to you.

Maker-Paul, 2025-04-01:
Hi thanks for answering. I would like to take the 3D chameleon into a cabinet design which is why I looked for the Tinkercad files. I can't understand why you don't want to make sure anything connected to 3DCameion is as good a quality as possible. It does after all reflect on the credibility of the project.

The embedded files are not the result of a lack of consolidation feature in Tinkercad, they are just the result of many previous separate edits not being tidied up by whoever was working on the files. As I said I am aware they are not the original files as they are marked as remixed. Once again I do now have a set of Mk4 Tinkercad files that do not have all the embedded copies if they are of use.

wjsteele, 2025-04-01:
_" I can't understand why you don't want to make sure anything connected to 3DCamelion is as good a quality as possible."_  Where did I say that???  

The files ON GITHUB are the latest and greatest files.  Which part of that isn't clear? They are the source for the 3DChameleon as it stands today and are accurate.  Again, if you have an issue with ANY of the files on Github, please let me know which ones and I'll fix them.  

The Tinkercad files are only for reference or if you want to mess with them.  They were the original source files from 7 years ago.  They've been edited and tweaked over the years... but the reference design for the 3DChameleon are the files in this Repository.  They're have been over 3,000 kits sold of that iteration, so they're known to be good working files.  I might add, all the previous versions of the 3DChameleon are somewhere in that history chain... like the Mk3.2, the Mk3.1, Mk3.0, Mk2.5 and the Mk2... they're all in that Tinkercad history.

Maker-Paul, 2025-04-02:
Hmmm, None ify what I've said is meant to be a criticism. Its meant as
constructive. If you have deliberatly kept redundent copies of objects
embedded in the model as some sort of archiveing method thats your choice.
The resulting files are huge so tinkercad struggles with alterations. But
so be it. If your online STL's are from these Tinkercad files then they
will be misaligned too. I will print some tommorrow and let you know. The
Tinkercad versions were out by approx 0.42mm probably not enough to stop it
working but the top and bottom of the main parts were visibly offset.

Nathan22211, 2025-06-22:
not 100% familiar with the internal workings of the 3d chameleon. But if I knew the basic concept behind its loading and unloading system, I might be able to recreate the models in onshape and get a STEP file for it. Arguably, the design is somewhat more compact than the 3MS but tougher to work with from my experience with the MK3.